hadess/rtl8723bs #96: Thanks? Status of upstreaming this?

hzulla, 2016-08-29:
Hi, thanks for this driver! Is there an effort underway to have it added to the Kernel tree? Would you consider adding a section to the readme that describes the status of this? Kind regards!

hadess, 2016-08-29:
It's not upstream, everything needs cleaning up.

closed 2016-08-29 by hadess

hzulla, 2016-08-29:
But is anybody working on a clean-up?

hadess, 2016-08-29:
It's being worked on, but it's unlikely to ever come to something mergeable. The driver is just too full of junk.

hzulla, 2016-08-29:
Ok, thanks for taking the time to response. What do you suggest can I do to learn about the progress of having Baytrail support in the mainline kernel? I'm trying to find out if I should keep my Baytrail device in the hope of coming better hardware support or sell if for something that's better supported in Linux.

hadess, 2016-08-29:
Read the archives of various distro and kernel mailing-lists. BayTrail devices are already the one-before-last generation of hardware, superseded by CherryTrail. BayTrail devices are in the $100-$200 range, new, so you might as well keep it until the support gets better.

hzulla, 2016-09-07:
Hey @lwfinger - I see that you have a [slightly updated copy of this driver](https://github.com/lwfinger/rtl8723bs) in your collection and mentioned this device some times on the Kernel mailing list. I'm not a kernel dev, but would really really love to have this thing in the mainline kernel. Can I help somehow with debugging / testing?