hadess/rtl8723bs #97: Question How to apply   patches from patches folder ?

Haxk20, 2016-09-02:
As title said but when i try "patch nameofpatch.patch"   it only show cursor in terminal and thats all i wait  for 5 minutes and nothing then i canceled it Please help

lwfinger, 2016-09-02:
Did you look at "man patch"? Patch reads from standard input, thus you need to redirect the input.

Haxk20, 2016-09-02:
Yes i look but when i type patch -p1 < 0001-mmc-sdhci-get-runtime-pm-when-sdio-irq-is-enabled.patch it give me this output:
can't find file to patch at input line 19
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From df5bb0eca11e6ab072908b6e0d2e7d9e01d1eb84 Mon Sep 17 00:00:00 2001
|From: Dong Aisheng <b29396@freescale.com>
|Date: Wed, 24 Jun 2015 21:28:48 +0200
|Subject: [PATCH] mmc: sdhci: get runtime pm when sdio irq is enabled
|
|SDIO cards may need clock to send the card interrupt to host.
|Thus, we get runtime pm when sdio irq is enabled to prevent the clock
|resource is released and put it when sdio irq is disabled.
|
|Signed-off-by: Dong Aisheng <b29396@freescale.com>
|---
| drivers/mmc/host/sdhci.c | 6 ++++--
| 1 file changed, 4 insertions(+), 2 deletions(-)
|
|diff --git a/drivers/mmc/host/sdhci.c b/drivers/mmc/host/sdhci.c
|index c80287a..f7433fa 100644
|--- a/drivers/mmc/host/sdhci.c
|+++ b/drivers/mmc/host/sdhci.c
--------------------------
File to patch:

lwfinger, 2016-09-02:
I do not know what directory you were in when you issued that command. The patch refers to "a/drivers/..." and you used "-p1", which would strip the first directory ("a/"). Thus if you had been in the directory of the kernel tree that contains "drivers", it would have worked.

Haxk20, 2016-09-02:
I was in the rtl8723bs-master/patches directory i was thinking that i need to be in the patches directory and if i can ask where is the kernel tree on ubuntu 16.04 LTS and do i need to download keenel source and recompile it with the patches ?

lwfinger, 2016-09-02:
No, you need to be in the tree that has the source file to be patched.

Obviously, you need to have the source that is to be patched. If you do not build the new kernel with the patches, they cannot be active.

Haxk20, 2016-09-02:
Ohhh ok i will download source tomorrow since its night and how much space compiling will take when i tryed it last tome everything went perfect for the moment when all 18.9 Gb run out and it stopped

speculatrix, 2016-09-02:
I think, Kaxk20, you really need to read some of the many good guides on building kernels and applying patches.

Haxk20, 2016-09-02:
Link ? "Yeah i know google is best friend but  im on mobile right now so small display not good for reading guides but really need +- 2 gb how much gb it will take ?

hadess, 2016-09-05:
@Haxk20 buy a book, there are plenty of cheap ones about Linux, and things like building a kernel. We can't handhold everyone trying to build a kernel, try asking your distributor about shipping the driver instead.

closed 2016-09-05 by hadess