hadess/rtl8723bs #98: Newer versions of the driver

lwfinger, 2016-09-05:
Anthony Wong at Canonical is investigating making this driver as part of Ubuntu's kernel. As a part of this activity, he has gotten newer drivers from Realtek. If anyone is interested in testing them, versions 4.3.16, 4.4.0 and 4.4.1 are available from https://github.com/anthonywong/rtl8723bs.

joerg-krause, 2016-09-05:
Nice! I guess asking about a Changelog is useless.

speculatrix, 2016-09-05:
it could be worse, it could be a Mediatek chip!

lwfinger, 2016-09-05:
I once asked Realtek for read access to their SVN repository. My request was ignored.

You could run diff on the various versions, but they have a habit of making a lot of trivial changes that obscure the important differences.

hzulla, 2016-09-07:
This is great news. Thanks @anthonywong!

hadess, 2017-04-12:
The driver is now upstream. Hopefully, Realtek will start cleaning the upstreamed driver instead of spinning messy driver updates. See commit 3bb1d33.

If you feel courageous, compare the newer versions of the driver to the original driver dump in this repository, and check whether the changes are still relevant to newer versions of the driver, upstream.

closed 2017-04-12 by hadess