hadess/rtl8723bs #100: Driver slows down time

nicman23, 2016-09-09:
time is slowed by 4x:
* `$ time ssh 192.168.1.120 sleep 2s`
```
real	0m9.008s
user	0m0.010s
sys	0m0.000s
```
* `$ time ssh 192.168.1.120 exit`
```
real	0m0.442s
user	0m0.007s
sys	0m0.000s
```
that ip being the tablet...

Using the patches in this [folder](https://github.com/nicman23/misc_aur/tree/master/linux-sp0-bytcr)
mostly sound and pmc clock and the patch from [here](https://github.com/hadess/rtl8723bs/issues/76#issuecomment-234706390) . This happens also without the sound and clock patches and only after connection.

lwfinger, 2016-09-10:
The most likely explanation is that the driver is locking out interrupts so that clock events are lost. The driver uses a lot of spin_lock() calls, and some of them might be replaced by mutex locking, which might help. The problem is that making such changes is dangerous when one does not have the hardware for testing. As you have both the hardware and the problem, you are in an ideal case for doing this kind of debugging.

nicman23, 2016-09-10:
And I'd agree with out but unfortunately i know nothing about writing drivers or any real language...

closed 2017-04-12 by hadess