hadess/rtl8723bs #103: rl8723bs monitor mode doesnt work

well0nez, 2016-09-19:
Hey Hadess, hey everyone.
I tried to install this driver (kernel 4.4 ubuntu 16.04) and use the monitor mode.
however, i cannot put the wlan0 device into monitor mode, it always fails due to unsupported mode.
when iam creating a new monitor device with iw, the creation is successfull, but capturing files with aircrack//tcpdump fails. nothing is captured.

could we try to figure out the reason?
i would help in any way, if you need access to the build, just ask, i can provide anything

iam using a wt8-b-102 toshiba tablet

well0nez, 2016-09-19:
i tried this driver too, but here, the monitor mode does not even get listed in iw infos
https://github.com/anthonywong/rtl8723bs

well0nez, 2016-09-19:
i tried this driver:
https://github.com/anthonywong/rtl8723bs

the monitor mode is not beeing listed, but setting the mode worked ootb, i got a kernel message, but system is still running.

lwfinger, 2016-09-19:
Are you saying that monitor mode works in the newer driver, which is what Anthony posted? If so, then anyone that wants monitor mode should be finding the necessary changes and post patches for this repo.

well0nez, 2016-09-19:
it was more than a mode that it is available there, but not working stable neither.
iwas referring to another issue where hadess said, monitor should be available if "we dont broke it".

so my main-thought is of course, to get this driver running with monitor mode.

closed 2017-04-12 by hadess