hadess/rtl8723bs #112: Acer One 10 - don't correct work

AngelOfMors, 2016-11-10:
Kali Linux (last full iso) Live USB (Persisctnce mode).

1. Some time later stop coming packets.
2. After connecting, you can not switch to another wifi network.
3. Low speed (it is necessary to recheck).

RussianNeuroMancer, 2016-11-14:
Did you test with patched kernel? 

As I understand there is latest version of required patch: https://github.com/burzumishi/linux-baytrail-flexx10/blob/master/kernel/patches/v4.8/0001-mmc_sdhci_on_intel_baytrail.patch
And there is deb-packages of patched kernel: https://yadi.sk/d/pW1Fbl-VyVPoR

AngelOfMors, 2016-11-14:
It's Live USB, don't patched.

RussianNeuroMancer, 2016-11-14:
This driver require patched kernel: https://github.com/hadess/rtl8723bs
> Note that on all those tablets, you will need to apply a few patches because to the BayTrail SDIO drivers

closed 2016-11-14 by AngelOfMors