hadess/rtl8723bs #113: Not working with kernel 4.8.x

tagorereddy, 2016-12-11:
I've been using the driver for a while now. It worked perfectly up to kernel 4.7.x but in kernel 4.8.x its failing.

DMESG:

```
[ 2993.528142] RTL8723BS: module init start
[ 2993.528147] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[ 2993.528149] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[ 2993.619588] RTL8723BS: rtw_ndev_init(wlan0)
[ 2993.621692] RTL8723BS: module init ret =0
[ 2993.680616] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2993.685093] rtl8723bs: accquire FW from file:rtlwifi/rtl8723bs_nic.bin
[ 2995.186605] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2995.298929] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2995.409729] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2995.517741] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2995.674250] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2995.839120] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2995.975658] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2996.099115] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2996.250988] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2996.398478] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
...
```
Goes on forever. 
The device is listed by iwconfig. But doesn't scan or any APs.

voobscout, 2016-12-20:
take patches from https://github.com/ferbar/rtl8723bs/tree/master/patches_4.7 it works for me on 4.8.13, also disable changing MAC on start:
```
$ cat /etc/NetworkManager/NetworkManager.conf
[device]
wifi.scan-rand-mac-address=no
```

tagorereddy, 2016-12-27:
The problem still persists. I think the current problem is due the latest kernel firmware 20161130. Please, someone confirm this by compiling the driver against latest kernel firmware.

janus-reith, 2016-12-27:
Same issue for me.
I rebuild my kernel 4.8.13 on arch with those two patches, set the sdio mode from acpi to pci and also added `wifi.scan-rand-mac-address=no`.
The bluetooth part is working now(with the start.sh from rtl8723bs_bt), but the wifi does not scan any APs.
Only releveant info in dmesg is several lines of `RTL8723BS: nolinked power save enter` and `leave`.

Output in iwconfig:
> wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
>           Mode:Ad-Hoc  Frequency=2.412 GHz  Cell: Not-Associated   
>           Sensitivity:0/0  
>           Retry:off   RTS thr:off   Fragment thr:off
>           Power Management:off
>           Link Quality:0  Signal level:0  Noise level:0
>           Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
>           Tx excessive retries:0  Invalid misc:0   Missed beacon:0

The output of `iw dev wlan0 scan` is empty.

Maybe OT, but I also hoped that those two patches you linked above would fix audio on my cherrytrail soc, but nothing changed...

janus-reith, 2016-12-28:
`sudo pacman -Ss linux-firmware` shows an older version,
core/linux-firmware 20161005.9c71af9-1th

Is still relevant as I compiled my own kernel which is more up to date?
Using the Arch Build System i generated -headers, -docs and the kernel itself.
So, not sure if firmware was affected as well.

If not, then I have the same issue on the older 20161005.

4octavio4, 2016-12-29:
I have the same problem with linuxium's kernel 4.4.0-53.74 , 4.8.0-30.32 and 4.9.0-8.9 and PCI mode in BIOS. I saw the same output than janus-reith with iwconfig and this error with dmesg:
IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

I can't see any AP with iwlist scan although the hardware is recognized and appears in /sys/bus/sdio/devices

tagorereddy, 2017-01-01:
**UPDATE:** I've managed to bring the device up by

nmcli c up <access_point_name>.

WiFi adapter connects to the access point and starts working albeit with less signal strength.
But after a minute or two, dmesg show following errors:

RTL8723BS: ERROR sd_recv_rxfifo: alloc recvbuf FAIL!

lwfinger, 2017-01-02:
You have either a very small RAM size, or there is a huge memory leak. In any case, the kernel has run out of memory that it can allocate..

tagorereddy, 2017-01-02:
@lwfinger I think both cases aren't true here. Although a memory leak is quite a possibility. However, one think is for sure. The driver is unstable with newer kernels. We'll know as more and more cases popup.

janus-reith, 2017-01-25:
nmcli c up <access_point_name> also does somethinf or me, at least network status icon changes to coonecting. But after some short time it fails to connect stating that it could not activate the connection.

I dont get the `RTL8723BS: ERROR sd_recv_rxfifo: alloc recvbuf FAIL! ` error in dmesg, it just adds another line of `RTL8723BS: nolinked power save enter` and `leave`

ferbar, 2017-03-01:
This kernel module was working for months on my tab. (Setup was in September 2016) Since the arch-linux update at the beginning of  2017 it is broken.
What I've tested:

use old 4.7 kernel which was working -> no success
downgrade linux-firmware to 20160730 which was working before -> no success
kill wpa_supplicant, connect to a passwordless wifi with `ipconfig wlan0 essid _name_` -> success

It seems that the problem is not linux-firmware related but wpa_supplicant or network-manager related. As long as wpa_supplicant is running `iwlist scanning` shows only a part of the available wifi networks or even nothing. Maybe the current version of wpa_supplicant is triggering a bug which results in a reset of the sta.

tagorereddy, 2017-03-01:
@ferbar I agree with your observation.

ferbar, 2017-03-01:
@lwfinger do you think it makes sense to enable the DBG_8192C macro in ioctl_cfg80211.c and check the differences between wpa_supplicant 1.2.6 and 1.2.5? I don't see any bad changes in the wpa_supplicant changelog (http://w1.fi/cgit/hostap/plain/wpa_supplicant/ChangeLog)

thx!

lwfinger, 2017-03-01:
I would likely start by capturing the over-the-air packets, and then enable the debug macros. If getting the transmitted packets is difficult, then seeing the contents of the transmitted and received packets would likely help.

One other approach would be to clone the hostapd/wpa_supplicant repo at  git://w1.fi/hostap.git and bisect between 1.2.5 and 1.2.6 to find the problem change.

mauzil, 2017-03-02:
Hi all.
I'm working on Giadatech NI-Z8300-UL board, which is equiped with rtl8723vau controller on SDIO bus.
I try the driver with kernel 4.9.2 but the system freezes when it scans for new  SSID.
Someone has experience with this situation?

tagorereddy, 2017-03-02:
@mauzil  You may need to install some SDIO/I2C related patches from Hans De Goede.

ferbar, 2017-03-02:
Update #2 for wpa_supplicant version 1.2.6:
`wpa_supplicant -u` = fail
`wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf` = working

ferbar, 2017-03-02:
@lwfinger: this is the wpa_supplicant commit which broke our kernel module

https://w1.fi/cgit/hostap/commit/?id=14220fe64c4e8180380d9fd15a985aa699b00e2f

wpa_supplicant -u -d gives a lot of output, however I don't understand the device - states, yet. But you can see that the device is sent to INTERFACE_DISABLED near scan lines. Have a look at the line

> Scan completed in 0.000276 seconds

An usb wifi takes 1.2 seconds to complete the scan.

tagorereddy, 2017-03-04:
I've also noticed that this driver is incompatible with NetworkManager's random MAC address assignment. So, adding the following to `/etc/NetworManager/NetworkManager.conf` should help.

```
[device]
wifi.scan-rand-mac-address=no
```

ferbar, 2017-03-05:
@tagorereddy I think this is not related with wpa_supplicant but a new issue :-)

closed 2017-04-12 by hadess