hadess/rtl8723bs #115: Yocto (Krogroth)

miss-wu, 2017-02-17:
Hi,

I'm trying to use this driver on Yocto (Krogroth). I was able to get the module to compile, however I receive an Oops when loading it on a CherryTrail device:

```
[    6.701951] BUG: unable to handle kernel paging request at 0000000400000004
[    6.710098] IP: [<ffffffffa019940d>] rtw_drv_register_netdev+0x3d/0x79 [r8723bs]
[    6.718836] PGD 7a4ec067 PUD 0 
[    6.722518] Oops: 0002 [#1] PREEMPT SMP 
[    6.727119] Modules linked in: intel_rapl(+) intel_powerclamp coretemp efivars(+) r8723bs(+) cfg80211 mei_txe mei processor_thermal_device dwc3_pci intel_soc_dts_iosf dw_dmac dw_dmac_core i2c_designware_platform i2c_designware_core hci_uart btintel spi_pxa2xx_platform pwm_lpss_platform pwm_lpss int3403_thermal int340x_thermal_zone int3400_thermal soc_button_array acpi_thermal_rel
[    6.766533] CPU: 3 PID: 160 Comm: udevd Not tainted 4.4.26-yocto-standard-custom #15
[    6.775726] Hardware name: Intel Corporation CHERRYVIEW D1 PLATFORM/Cherry Trail CR, BIOS CH2TCR.X64.0005.R24.1605180428 05/18/2016
[    6.789434] task: ffff88007a5a1f80 ti: ffff88007a698000 task.ti: ffff88007a698000
[    6.798103] RIP: 0010:[<ffffffffa019940d>]  [<ffffffffa019940d>] rtw_drv_register_netdev+0x3d/0x79 [r8723bs]
[    6.809651] RSP: 0018:ffff88007a69bb38  EFLAGS: 00010246
[    6.815807] RAX: 0000000400000004 RBX: ffff88007a8a4000 RCX: 0000000000000000
[    6.824071] RDX: 000000000c70f35c RSI: ffff88007a69baf5 RDI: ffff88007a8a4000
[    6.832332] RBP: ffff88007a69bb48 R08: 000000000000000a R09: 000000000000fffb
[    6.840594] R10: ffffffff81b8f1c8 R11: ffff88007a69baf5 R12: ffffc90000606000
[    6.848857] R13: ffff88007abb7800 R14: ffff88007a8a4000 R15: ffffffffa01d15e0
[    6.857121] FS:  00007f8a4030a780(0000) GS:ffff880079d80000(0000) knlGS:0000000000000000
[    6.866484] CS:  0010 DS: 0000 ES: 0000 CR0: 000000008005003b
[    6.873134] CR2: 0000000400000004 CR3: 000000007a683000 CR4: 00000000001006e0
[    6.881384] Stack:
[    6.883716]  ffff88007a837800 ffffc90000606000 ffff88007a69bb78 ffffffffa019a322
[    6.892347]  ffff88007abb7808 ffff88007abb7800 0000000000000000 ffffffffa01df380
[    6.900983]  ffff88007a69bbb0 ffffffff8178a0bd ffff88007abb7808 0000000000000000
[    6.909607] Call Trace:
[    6.912574]  [<ffffffffa019a322>] rtw_drv_init+0x1f8/0x2c7 [r8723bs]
[    6.919957]  [<ffffffff8178a0bd>] sdio_bus_probe+0xed/0x130
[    6.926426]  [<ffffffff815adcf7>] driver_probe_device+0x1f7/0x310
[    6.933492]  [<ffffffff815ade94>] __driver_attach+0x84/0x90
[    6.939954]  [<ffffffff815ade10>] ? driver_probe_device+0x310/0x310
[    6.947218]  [<ffffffff815abda4>] bus_for_each_dev+0x64/0xa0
[    6.953780]  [<ffffffff815ad5be>] driver_attach+0x1e/0x20
[    6.960040]  [<ffffffff815ad1f1>] bus_add_driver+0x1b1/0x220
[    6.966600]  [<ffffffffa01fd000>] ? 0xffffffffa01fd000
[    6.972561]  [<ffffffff815ae7a0>] driver_register+0x60/0xe0
[    6.979025]  [<ffffffff81789ec1>] sdio_register_driver+0x31/0x40
[    6.986079]  [<ffffffffa01fd05b>] rtw_drv_entry+0x5b/0x91 [r8723bs]
[    6.993353]  [<ffffffff810003b6>] do_one_initcall+0x86/0x1a0
[    6.999919]  [<ffffffff81134e98>] do_init_module+0x5f/0x1c3
[    7.006385]  [<ffffffff810e099d>] load_module+0x1bed/0x21f0
[    7.012849]  [<ffffffff810dd8a0>] ? __symbol_put+0x60/0x60
[    7.019214]  [<ffffffff810ddbdb>] ? copy_module_from_fd.isra.58+0xdb/0x130
[    7.027184]  [<ffffffff810e115f>] SYSC_finit_module+0x8f/0xa0
[    7.033845]  [<ffffffff810e118e>] SyS_finit_module+0xe/0x10
[    7.040296]  [<ffffffff8197d1d7>] entry_SYSCALL_64_fastpath+0x12/0x66
[    7.047756] Code: a0 00 02 00 00 49 8b 9c 24 50 40 00 00 48 8d b7 36 3b 00 00 48 89 df e8 6d f6 ff ff 48 8b 83 10 03 00 00 41 8b 94 24 5c 3b 00 00 <89> 10 41 0f b7 94 24 60 3b 00 00 66 89 50 04 48 89 df e8 4c 52 
[    7.070852] RIP  [<ffffffffa019940d>] rtw_drv_register_netdev+0x3d/0x79 [r8723bs]
[    7.079673]  RSP <ffff88007a69bb38>
[    7.083715] CR2: 0000000400000004
[    7.201204] ---[ end trace 27dbe359879ef5f5 ]---
```

I've narrowed down:
`RIP  [<ffffffffa019940d>] rtw_drv_register_netdev+0x3d/0x79 [r8723bs]`
to os_dep/os_intfs.c#930:
`memcpy(pnetdev->dev_addr, padapter->eeprompriv.mac_addr, ETH_ALEN);`

I'm able to output `padapter->eeprompriv.mac_addr`:
5c:f3:70:0c:8e:1e
and `ETH_ALLEN`:
6

Any ideas on where I can go from here?

Thank you for your time!

lwfinger, 2017-02-18:
Please verify the location that fails by changing directory to the source and entering the following:

gdb r8723bs.ko

At the gdb prompt, enter "list *rtw_drv_register_netdev+0x3d" and post the output.

The reason I want you to do the above is that I got a different location for the error than you are referencing.

miss-wu, 2017-02-21:
Hi,

Here is the output:

> gdb ./r8723bs.ko
> GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
> Copyright (C) 2014 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> and "show warranty" for details.
> This GDB was configured as "x86_64-linux-gnu".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <http://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
> <http://www.gnu.org/software/gdb/documentation/>.
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from ./r8723bs.ko...done.
> (gdb) list *rtw_drv_register_netdev+0x3d
> 0x27431 is in rtw_drv_register_netdev (/data/n27x_krogoth_01182017/poky/build/tmp/work-shared/cht/kernel-source/drivers/net/wireless/realtek/rtlwifi/cht/os_dep/os_intfs.c:930).
> 925             struct net_device *pnetdev = padapter->pnetdev;
> 926
> 927             /* alloc netdev name */
> 928             rtw_init_netdev_name(pnetdev, name);
> 929
> 930             memcpy(pnetdev->dev_addr, padapter->eeprompriv.mac_addr, ETH_ALEN);
> 931
> 932             /* Tell the network stack we exist */
> 933             if (register_netdev(pnetdev) != 0) {
> 934                     DBG_871X(FUNC_NDEV_FMT "Failed!\n", FUNC_NDEV_ARG(pnetdev));
> (gdb)

Please let me know if there is any other information I can provide.

Thank you!

lwfinger, 2017-02-21:
Strange.  The attached patch file will give some extra info. Apply it with

patch -p1 < check_oops.txt

Rebuild and reinstall

[check_oops.txt](https://github.com/hadess/rtl8723bs/files/791747/check_oops.txt)

lwfinger, 2017-02-21:
[check_oops.txt](https://github.com/hadess/rtl8723bs/files/791754/check_oops.txt)

First version was faulty. Use this one.

miss-wu, 2017-02-21:
Hi,

This is the output:
>[    6.569456] pnetdev->dev_addr 0000000400000004
>[    6.578566] padapter->eeprompriv.mac_addr ffffc90000709b5c

Thank you!

miss-wu, 2017-02-22:
Hi,

I'm probably wrong/going about this wrong so please let me know if I am.

**Verified 'padapter->eeprompriv.mac_addr' and 'ETH_ALEN'**
I created a test pointer and was able to do a memcpy with 'padapter->eeprompriv.mac_addr' and 'ETH_ALEN':
```
    char *test = rtw_zmalloc( 6 * sizeof(char) );
    memcpy(test, padapter->eeprompriv.mac_addr, ETH_ALEN);
```
which provides:
> 	MAC_ARG(test) = 5c:f3:70:0c:8e:1e
> 	MAC_ARG(padapter->eeprompriv.mac_addr) = 5c:f3:70:0c:8e:1e
> 	ETH_ALEN = 6

**Do you know where 'pnetdev->dev_addr' is initialized?** 
By working my way backwards using `pr_info("pnetdev->dev_addr %p\n", pnetdev->dev_addr);` I think I found where it is allocated: 

- `pr_info("pnetdev->dev_addr %p\n", pnetdev->dev_addr);` stopped working before `pnetdev = rtw_init_netdev(padapter);` in [os_dep/sdio_intf.c#L347](https://github.com/hadess/rtl8723bs/blob/275ece719635a9304d75ccaa6ce994e87774cb54/os_dep/sdio_intf.c#L347)

- [`pnetdev = rtw_alloc_etherdev_with_old_priv(sizeof(struct adapter), (void *)old_padapter);`](https://github.com/hadess/rtl8723bs/blob/275ece719635a9304d75ccaa6ce994e87774cb54/os_dep/os_intfs.c#L550)

- [`pnetdev = alloc_etherdev_mq(sizeof(struct rtw_netdev_priv_indicator), 4);`](https://github.com/hadess/rtl8723bs/blob/275ece719635a9304d75ccaa6ce994e87774cb54/os_dep/osdep_service.c#L236)

However, I haven't found where dev_addr is initialized, do you know where this happens?

Thank you!

lwfinger, 2017-02-22:
I think you are doing it correctly.
From your output,
    [ 6.569456] pnetdev->dev_addr 0000000400000004
    [ 6.578566] padapter->eeprompriv.mac_addr ffffc90000709b5c
I knew that pnetdev->dev_addr was wrong, as a virtual address should be a large number such as the mac_addr.

I'm not sure what is going wrong, but I did find a place where the error returns were all screwed up. Please do a git pull and report if you get a new error message. Thanks.

miss-wu, 2017-02-22:
Hi,

After updating I didn't see a different error.
```
[    5.888038] RTL8723BS: module init start
[    5.893112] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[    5.901060] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[    6.132334] pnetdev->dev_addr 0000000400000004
[    6.137400] padapter->eeprompriv.mac_addr ffffc900003fdb5c
[    6.143613] BUG: unable to handle kernel paging request at 0000000400000004
[    6.151749] IP: [<ffffffffa0166461>] rtw_drv_register_netdev+0x70/0xb3 [r8723bs]
[    6.160467] PGD 7a7cf067 PUD 0
[    6.164148] Oops: 0002 [#1] PREEMPT SMP
[    6.168749] Modules linked in: mei_txe mei dwc3_pci processor_thermal_device intel_soc_dts_iosf r8723bs(+) dw_dmac dw_dmac_core cfg80211 i2c_designware_platform(+) i2c_designware_core spi_pxa2xx_platform hci_uart btintel pwm_lpss_platform pwm_lpss int3403_thermal int3400_thermal int340x_thermal_zone acpi_thermal_rel soc_button_array
[    6.203260] CPU: 0 PID: 172 Comm: udevd Not tainted 4.4.26-yocto-standard-custom #1
[    6.212127] Hardware name: Intel Corporation CHERRYVIEW D1 PLATFORM/Cherry Trail CR, BIOS CH2TCR.X64.0005.R24.1605180428 05/18/2016
[    6.225812] task: ffff88007a75b480 ti: ffff88007a7dc000 task.ti: ffff88007a7dc000
[    6.234459] RIP: 0010:[<ffffffffa0166461>]  [<ffffffffa0166461>] rtw_drv_register_netdev+0x70/0xb3 [r8723bs]
[    6.245983] RSP: 0018:ffff88007a7dfb30  EFLAGS: 00010286
[    6.252132] RAX: 0000000400000004 RBX: 0000000000000000 RCX: 0000000000000000
[    6.260387] RDX: 000000000c70f35c RSI: 0000000000000246 RDI: ffffffff81f6a328
[    6.268641] RBP: ffff88007a7dfb50 R08: 0000000000000000 R09: 000000000000025d
[    6.276893] R10: ffffffff81b8f238 R11: 000000000000025d R12: ffff88007a564000
[    6.285144] R13: ffffc900003fa000 R14: ffffc900003fdb5c R15: ffffffffa019e680
[    6.293399] FS:  00007fb20a8bf780(0000) GS:ffff880079a00000(0000) knlGS:0000000000000000
[    6.302758] CS:  0010 DS: 0000 ES: 0000 CR0: 000000008005003b
[    6.309407] CR2: 0000000400000004 CR3: 000000007a7ce000 CR4: 00000000001006f0
[    6.317659] Stack:
[    6.319992]  ffff88007974b000 ffffc900003fa000 ffff88007aa5ec00 ffff88007a564000
[    6.328623]  ffff88007a7dfb80 ffffffffa016737d ffff88007aa5ec08 ffff88007aa5ec00
[    6.337253]  0000000000000000 ffffffffa01ac480 ffff88007a7dfbb8 ffffffff8178a17d
[    6.345884] Call Trace:
[    6.348838]  [<ffffffffa016737d>] rtw_drv_init+0x1f8/0x2c7 [r8723bs]
[    6.356207]  [<ffffffff8178a17d>] sdio_bus_probe+0xed/0x130
[    6.362670]  [<ffffffff815add57>] driver_probe_device+0x1f7/0x310
[    6.369731]  [<ffffffff815adef4>] __driver_attach+0x84/0x90
[    6.376187]  [<ffffffff815ade70>] ? driver_probe_device+0x310/0x310
[    6.383450]  [<ffffffff815abe04>] bus_for_each_dev+0x64/0xa0
[    6.390006]  [<ffffffff815ad61e>] driver_attach+0x1e/0x20
[    6.396259]  [<ffffffff815ad251>] bus_add_driver+0x1b1/0x220
[    6.402814]  [<ffffffffa01ca000>] ? 0xffffffffa01ca000
[    6.408769]  [<ffffffff815ae800>] driver_register+0x60/0xe0
[    6.415224]  [<ffffffff81789f80>] sdio_register_driver+0x20/0x30
[    6.422258]  [<ffffffffa01ca05b>] rtw_drv_entry+0x5b/0x91 [r8723bs]
[    6.429519]  [<ffffffff810003b6>] do_one_initcall+0x86/0x1a0
[    6.436075]  [<ffffffff81170e3f>] ? __vunmap+0xbf/0xe0
[    6.442032]  [<ffffffff811830ec>] ? kfree+0x16c/0x180
[    6.447889]  [<ffffffff81134f08>] do_init_module+0x5f/0x1c3
[    6.454347]  [<ffffffff810e0a0d>] load_module+0x1bed/0x21f0
[    6.460804]  [<ffffffff810dd910>] ? __symbol_put+0x60/0x60
[    6.467154]  [<ffffffff810ddc4b>] ? copy_module_from_fd.isra.58+0xdb/0x130
[    6.475115]  [<ffffffff810e11cf>] SYSC_finit_module+0x8f/0xa0
[    6.490745]  [<ffffffff810e11fe>] SyS_finit_module+0xe/0x10
[    6.506194]  [<ffffffff8197d517>] entry_SYSCALL_64_fastpath+0x12/0x66
[    6.522556] Code: a0 e8 b1 e8 fc e0 4d 8d b5 5c 3b 00 00 4c 89 f6 48 c7 c7 00 7d 1a a0 e8 9b e8 fc e0 49 8b 84 24 10 03 00 00 41 8b 95 5c 3b 00 00 <89> 10 41 0f b7 95 60 3b 00 00 66 89 50 04 4c 89 e7 e8 39 85 6b
[    6.563861] RIP  [<ffffffffa0166461>] rtw_drv_register_netdev+0x70/0xb3 [r8723bs]
[    6.581985]  RSP <ffff88007a7dfb30>
[    6.595222] CR2: 0000000400000004
[    6.646622] ---[ end trace fa8976c20ad6afa7 ]---
```

```
gdb ./tmp/work/corei7-64-intel-common-poky-linux/linux-yocto/4.4.26+gitAUTOINC+4bfb093467-r0/image/lib/modules/4.4.26-yocto-standard-custom/kernel/drivers/net/wireless/realtek/rtlwifi/cht/r8723bs.ko
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
[... 7 lines cut ...]
Reading symbols from ./tmp/work/corei7-64-intel-common-poky-linux/linux-yocto/4.4.26+gitAUTOINC+4bfb093467-r0/image/lib/modules/4.4.26-yocto-standard-custom/kernel/drivers/net/wireless/realtek/rtlwifi/cht/r8723bs.ko...done.
(gdb) list *rtw_drv_register_netdev+0x70
0x27485 is in rtw_drv_register_netdev (/data/n27x_krogoth_01182017/poky/build/tmp/work-shared/cht-t3/kernel-source/drivers/net/wireless/realtek/rtlwifi/cht/os_dep/os_intfs.c:935).
930                     return _FAIL;
931
932             pr_info("pnetdev->dev_addr %p\n", pnetdev->dev_addr);
933             pr_info("padapter->eeprompriv.mac_addr %p\n", padapter->eeprompriv.mac_addr);
934
935             memcpy(pnetdev->dev_addr, padapter->eeprompriv.mac_addr, ETH_ALEN);
936
937             /* Tell the network stack we exist */
938             if (register_netdev(pnetdev) != 0) {
939                     DBG_871X(FUNC_NDEV_FMT "Failed!\n", FUNC_NDEV_ARG(pnetdev));
(gdb) 
```

```
/ #
/ # dmesg | grep dev_alloc_name
/ #
```

Please let me know if there is any other information that I can provide.

Thank you!

lwfinger, 2017-02-26:
I do not see anything yet. I added a dump of pnetdev after its allocation in rtw_init_netdev(). Pull, build and install, then 'dmesg | grep pnetdev'.

closed 2017-04-12 by hadess