hadess/rtl8723bs #118: rtl8723bs not build against 4.11rc1

RussianNeuroMancer, 2017-03-06:
Hello!
rtl8723bs not build against 4.11rc1. Error message:

```
In file included from /var/lib/dkms/rtl8723bs/0.1.5~ubuntu1/build/include/drv_types.h:27:0,
                 from /var/lib/dkms/rtl8723bs/0.1.5~ubuntu1/build/core/rtw_cmd.c:17:
/var/lib/dkms/rtl8723bs/0.1.5~ubuntu1/build/include/osdep_service.h: In function ‘thread_enter’:
/var/lib/dkms/rtl8723bs/0.1.5~ubuntu1/build/include/osdep_service.h:125:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
/var/lib/dkms/rtl8723bs/0.1.5~ubuntu1/build/include/osdep_service.h: In function ‘flush_signals_thread’:
/var/lib/dkms/rtl8723bs/0.1.5~ubuntu1/build/include/osdep_service.h:130:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
/var/lib/dkms/rtl8723bs/0.1.5~ubuntu1/build/include/osdep_service.h:132:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
cc1: all warnings being treated as errors

```

(0.1.5 is version of rtl8723bs package; I tried to use latest rtl8723bs sources, but outcome is the same.)

hadess, 2017-03-06:
See #117.

closed 2017-03-06 by hadess

lwfinger, 2017-03-06:
The patch from #117 never made it into the code. I just pushed the fix.

RussianNeuroMancer, 2017-03-06:
Thanks! it's working now, so I will update package in [PPA](https://launchpad.net/~russianneuromancer/+archive/ubuntu/drivers) tomorrow.

RussianNeuroMancer, 2017-03-07:
Latest sources is published in PPA [ppa:russianneuromancer/drivers](https://launchpad.net/~russianneuromancer/+archive/ubuntu/drivers)

crank123, 2017-03-08:
Thank goodness it's patched. I literally just came to github to report my build problems on 4.11rc1.