hadess/rtl8723bs #121: Kernel patch links don't work

flarn2006, 2017-03-22:
This driver requires kernel patches in order to function, but the links in the readme for the patches are dead. Because of this I am unable to use the driver. Could you please update the links, perhaps even upload the patches to the repository?

Thank you.

RussianNeuroMancer, 2017-03-24:
Updated links would be nice, meanwhile you can find patch [here](https://github.com/hadess/rtl8723bs/issues/76#issuecomment-272932161)

ferbar, 2017-03-29:
Have a look at my pull request #109

Laszlo-Fiat, 2017-04-02:
Well, the necessary patch is on it's way to mainline linux, thanks to Hans de Goede. 

It has now landed in linux-next: https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/mmc?id=923713b357455cfb9aca2cd3429cb0806a724ed2

According to this: https://www.spinics.net/lists/linux-mmc/msg43455.html it will be added to newly released stable kernels versions.

braiam, 2017-04-04:
@Laszlo-Fiat pure curiosity, what were the other two patches for?

Laszlo-Fiat, 2017-04-10:
Good news, the last missing kernel patch has made it's way to linux 4.10.9 and linux 4.9.21.
So, linux 4.11-rc5, linux 4.10.9, linux 4.9.21 and later versions will not need any externel patches to be able to compile and use rtl8723bs.

@braiam if you ask about the other patches in the rtl8723bs patches directory, that functionality has already been upstreamed by Adrian Hunter quite some time ago, only this SDIO IRQ patch was missing from mainline. You can read the comments of the patches in them.

closed 2017-04-12 by hadess