hadess/rtl8723bs #123: Linux A20-OLinuXino 3.4.103-00033-g9a1cd03-dirty

Adportas, 2017-03-31:
Hello
Is there a way to build this software in kernel Linux A20-OLinuXino 3.4.103-00033-g9a1cd03-dirty
We get this error message at compile time:
/home/daniel/Proyectos/A20/rtl8723bs/include/ioctl_cfg80211.h:21:2: error: #error This driver only works with kernel 3.19 and later
Thank's in advance
Daniel
[image]

hadess, 2017-03-31:
No, the error message says it all, you need a newer kernel.

closed 2017-03-31 by hadess

Adportas, 2017-03-31:
Thanks for your response, do you know of any Kernel supporting A20 flashable from NAND that I could use for this? I can’t find anything beyond 3.4.x for A20 that supports both NAND and this wi-fi module.

Adportas, 2017-03-31:
We are working with an A20(an ARM based SOC) SOM with sunxi linux where all the used kernels are 3.4.X and i don't find any with version 3.19.* available, can you give some tips on this, I’am not familiarized with the procedure to satisfy the required version in this scenario